denkitronik/ADS1118 #6: ESP8266 Support ??

ianburton20, 2020-03-23:
Might this library work on an ESP8266? Anyone used it on an ESP8266?

alvaro-salazar, 2020-04-09:
It's possible, now I am working on that.

ianburton20, 2020-04-09:
Great if the ADS1118 could be used with the ESP8266. Using an SPI bus may well speed up sampling. With an ADS1115 connected to the ESP8266 via I2C, the best I can do is sample every 2-3mS. The ADS1115/8 is capable of sampling every 1.1mS (approx).

QuitButton, 2020-05-23:
I can't get the examples to compile when the target is ESP8266:

```
WARNING: library ADS1118 claims to run on avr, esp32 architecture(s) and may be incompatible with your current board which runs on esp8266 architecture(s).
basicExampleAds1118:17:19: error: no matching function for call to 'ADS1118::ADS1118(int)'

 ADS1118 ads1118(CS);

                   ^

C:\...\ADS1118\examples\basicExampleAds1118\basicExampleAds1118.ino:17:19: note: candidates are:

In file included from C:\...\ADS1118\examples\basicExampleAds1118\basicExampleAds1118.ino:10:0:

C:\...\Arduino\libraries\ADS1118/ADS1118.h:38:7: note: ADS1118::ADS1118()

 class ADS1118 {

       ^

C:\...\libraries\ADS1118/ADS1118.h:38:7: note:   candidate expects 0 arguments, 1 provided

C:\...\libraries\ADS1118/ADS1118.h:38:7: note: constexpr ADS1118::ADS1118(const ADS1118&)

C:\...\libraries\ADS1118/ADS1118.h:38:7: note:   no known conversion for argument 1 from 'int' to 'const ADS1118&'

C:\...\libraries\ADS1118/ADS1118.h:38:7: note: constexpr ADS1118::ADS1118(ADS1118&&)

C:\...\libraries\ADS1118/ADS1118.h:38:7: note:   no known conversion for argument 1 from 'int' to 'ADS1118&&'

exit status 1
no matching function for call to 'ADS1118::ADS1118(int)'
```

Mcharlsto, 2021-10-31:
Hi. Is there any update on this? I'm still getting the same error above when trying to compile.

laercionit, 2023-11-11:
Hello, I would like guidance on adapting the LIB to esp8266.

j-p-jadeja, 2026-10-07:
@alvaro-salazar any progress on this? If you are busy, could you help or guide me to achieve this.

alvaro-salazar, 2026-10-07:
Hi! Due to work commitments, I haven't had time to maintain this project, but I definitely plan to work on it again. Porting the code to other architectures is on the to-do list. If you'd like to give it a shot, I can guide you through it if you need it done soon. Specifically, are you looking to port it to the ESP8266 architecture?

j-p-jadeja, 2026-10-07:
Yes, I have a Adafruit Feather HUZZAH ESP8266 which I need to get working for a project.