denkitronik/ADS1118 #12: Support for Teensy Microcontrollers

chrismailer, 2021-09-22:
Great library, thanks a bunch! Just wanted to suggest a quick change which would add support for the Teensy Microcontrollers. Would just need to change
```
#if defined(__AVR__)
```
to
```
#if defined(__AVR__) || defined(CORE_TEENSY)
```
tested and everything should work as expected.

Fluxanode, 2022-04-27:
Was this change made in the source or does it still need to be changed?  Where exactly is it changed?

Fluxanode, 2023-06-30:
hello?

Fluxanode, 2023-06-30:
Trying to compile for the teensy 4.1 and seeing these error messages:

D:\ChuckW\Documents\Microcontrollers\IC's modules and discreets\ADS1118\ADS1118_library-1.0.3\ADS1118_library-1.0.3\examples\ads1118example\ads1118example.ino:17:19: error: no matching function for call to 'ADS1118::ADS1118(int)'
   17 | ADS1118 ads1118(CS);
      |                   ^
In file included from D:\ChuckW\Documents\Microcontrollers\IC's modules and discreets\ADS1118\ADS1118_library-1.0.3\ADS1118_library-1.0.3\examples\ads1118example\ads1118example.ino:10:
d:\ChuckW\Arduino\libraries\ADS1118_library/ADS1118.h:38:7: note: candidate: 'ADS1118::ADS1118()'
   38 | class ADS1118 {
      |       ^~~~~~~
d:\ChuckW\Arduino\libraries\ADS1118_library/ADS1118.h:38:7: note:   candidate expects 0 arguments, 1 provided
d:\ChuckW\Arduino\libraries\ADS1118_library/ADS1118.h:38:7: note: candidate: 'constexpr ADS1118::ADS1118(const ADS1118&)'
d:\ChuckW\Arduino\libraries\ADS1118_library/ADS1118.h:38:7: note:   no known conversion for argument 1 from 'int' to 'const ADS1118&'
d:\ChuckW\Arduino\libraries\ADS1118_library/ADS1118.h:38:7: note: candidate: 'constexpr ADS1118::ADS1118(ADS1118&&)'
d:\ChuckW\Arduino\libraries\ADS1118_library/ADS1118.h:38:7: note:   no known conversion for argument 1 from 'int' to 'ADS1118&&'

Using library ADS1118 library at version 1.0.3 in folder: D:\ChuckW\Arduino\libraries\ADS1118_library 
Using library SPI at version 1.0 in folder: C:\Users\ChuckW\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.58.1\libraries\SPI 
exit status 1

Compilation error: no matching function for call to 'ADS1118::ADS1118(int)'

Can you give me an idea why?

Fluxanode, 2023-06-30:
@chrismailer what teensy did you get this working with?  I can't get it to compile even with your changes.  I changed 4 lines in ADS118.cpp from #if defined(__AVR__) to #if defined(__AVR__) || defined(CORE_TEENSY) on lines 68, 88, 185 and 269.  Still won't compile see my post above.  Any help appreciated.

Fluxanode, 2023-06-30:
Problem solved, I missed the line in the .h file.  Moving on to testing with teensy 4.1