denkitronik/ADS1118 #14: Function arguments

Fluxanode, 2021-12-28:
What are the function arguments to the begin() function?  without dissecting the code i want to use single ended mode not the differential mode used in the examples.  In fact a completed list for all functions would be helpful.

Fluxanode, 2022-04-27:
Can I get an answer?  DIFFERENTIAL AIN0-AIN1 to SINGLE AIN0-AIN1 maybe?

Fluxanode, 2022-04-28:
HELP!!!

alvaro-salazar, 2022-04-28:
Hi @Fluxanode, unfortunately begin method has not any argument to change the input. But there's a specific method to do it:

(This is part of the example file: examples/ads1118example/ads1118example.ino)

/* Changing the input selected. 
       Available values: Diferential inputs: DIFF_0_1, DIFF_0_3, DIFF_1_3, DIFF_2_3. 
                                  Single ended input: AIN_0, AIN_1, AIN_2, AIN_3*/
ads1118.setInputSelected(ads1118.AIN_0);             //Using the setter method to change the input selected

A complete documentation is in the folder docs, in html format and in latex format (it is used to generate a pdf file). In the next version I will add the documentation in pdf and in the README.

I hope this answer to your question.

Fluxanode, 2022-04-29:
Thanks.  Also I’m using the teensy 4.1 and I was wondering about the answer to issue Support for Teensy Microcontrollers issue #12

 

Has this been implemented? 

 

 

Chuck Weaver

SideLobe Inc.

(281) 210-7517

alvaro-salazar, 2022-04-30:
Hi Chuck, I haven't implemented the issue #12 but, I'm going to try to do it this weekend. Could you test the implementation in your Teensy please? I am not able to check it because I dont have a Teensy, and it would take a week to get one. 
I will put the commit this weekend, or if you want I can send you the files.

Alvaro Salazar
Denkitronik SAS

Fluxanode, 2022-04-30:
Maybe next week, I'm not near my office for the next few days. But you can send me your files and when I can I'll set up a new test.

Fluxanode, 2023-06-29:
Alvaro Salazar, I'm back on this project again,  Can you be more specific about where to make the changes you refer too?  Also did you make the changes referred to in ssue 12 https://github.com/denkitronik/ADS1118/issues/12?  I can test it for you and or where in he code do the changes need to be made?

Fluxanode, 2023-07-07:
Tested on Teensy 4.1 successfully